kash4kev/vscode-esp8266fs #2: Upload issue "Can't determine FlashMode"

damianog, 2018-05-26:
Since last update when I try to upload my data folder the extension tells: "Can't determine FlashMode"

kash4kev, 2018-05-26:
Hi damianog,

What's in your arduino.json file?

damianog, 2018-05-26:
Hi,

```javascript
{
   "board": "esp8266:esp8266:nodemcuv2",
   "configuration": "CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200",
   "port": "COM3",
   "sketch": "EVRemote.ino",
   "output": "../build"
}
```

kash4kev, 2018-05-26:
Works fine on my machine with that board/config (under Windows):

Found ESP8266 packages: C:\Users\Kev\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
--- Uploading SPIFFS file with esptool[.exe] ---
SPIFFS Uploading Image... (c:\Code\PlatformIO\Blink\spiffs.image)
  [SPIFFS] program: C:\Users\Kev\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13\esptool.exe
  [SPIFFS] address: 0x300000
  [SPIFFS] reset  : nodemcu
  [SPIFFS] port   : COM18
  [SPIFFS] speed  : 115200

---

What OS are you running?  Can you add "esp8266fs.logLevel": "verbose" to your settings and see if that gives us a better clue as to where it's screwing up.

Thanks, Kev

damianog, 2018-05-26:
Hi @kash4kev 

If I add "FlashMode=keep,FlashFreq=keep" in configuration

`"configuration": "CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200,FlashMode=keep,FlashFreq=keep",`

I managed to start the upload process.

I used "keep" as default parameter as explained in https://github.com/espressif/esptool

But I can't set these pararameters by board configuration

[image]

Maybe you should pass these parameters by default if absent or avoid FlashMode and FlashFreq arguments if not defined.

Thanks.

TildenG, 2018-10-28:
I'm having the same problem and even after using your fix it freezes before It starts to pack the file...

ESP8266 SPIFFS interface started: "pack".
Preferences Path: "C:\Users\FirstName LastName\AppData\Local\Arduino15"
Reading preferences from "C:\Users\FirstName LastName\AppData\Local\Arduino15\preferences.txt"
ESP8266 Data Files path: "c:\Users\FirstName LastName\OneDrive\Documents_\Arduino\testing\FH_Tbot\data"
SPIFFS Image: "c:\Users\FirstName LastName\OneDrive\Documents_\Arduino\testing\FH_Tbot\spiffs.image"
Arduino User Path: "C:\Users\FirstName LastName\Documents\Arduino"
Found ESP8266 packages: C:\Users\FirstName LastName\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Tools Path: "C:\Users\FirstName LastName\AppData\Local\Arduino15\packages\esp8266\tools"
Output Port: COM4
--- Packing SPIFFS file ---
SPIFFS Creating Image... (c:\Users\FirstName LastName\OneDrive\Documents_\Arduino\testing\FH_Tbot\spiffs.image)
  [SPIFFS] program: C:\Users\FirstName LastName\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0\mkspiffs.exe
  [SPIFFS] data   : c:\Users\FirstName LastName\OneDrive\Documents_\Arduino\testing\FH_Tbot\data
  [SPIFFS] size   : 3053K
  [SPIFFS] page   : 256
  [SPIFFS] block  : 8192
Running: "C:/Users/FirstName LastName/AppData/Local/Arduino15/packages/esp8266/tools/mkspiffs/0.2.0/mkspiffs.exe" --create "c:/Users/FirstName LastName/OneDrive/Documents_/Arduino/testing/FH_Tbot/data" --size 3125248 --page 256 --block 8192 "c:/Users/FirstName LastName/OneDrive/Documents_/Arduino/testing/FH_Tbot/spiffs.image"  

And.... nothing, it just sits there, after leaving it for some time if I try to pack it again it says "ESP8266FS already running "pack"! ".

I'm running this on Windows 10.

TildenG, 2018-10-30:
I relocated esptool.exe and mkspiffs.exe to a directory with no spaces in there name, updated there location in the visual studio ESP8266FS settings.

Relocated my project to a directory with no spaces and its working fine.

rossiniscarface, 2018-11-13:
Yeah, had also a whitespace problem on my environment. 
My project folder had a space in it. 
Result: files to pack not found, image not found. If i enable verbose logging i can copy the comands and execute them without problems.