kash4kev/vscode-esp8266fs #3: Help with "one-button" dev cycle

josemariaaraujo, 2018-07-04:
In the readme you say:
> Tip: Add the "esp866fs.packSpiffs" and "esp866fs.uploadSpiffs" command to your gulp/webpack toolchain to turn your VSCode/ESP8266/ESP32 into a "one-button" dev cycle.

I'm new to vscode and the searches I've made came empty regarding on how to do this. Can you please write instructions on how to do it?
Thanks!

kash4kev, 2018-07-04:
Hey Jose,

It's the 4th here in the States, so I can't really get back to you
immediately, but I promise I'll write something up for ya and help you
through it.

Cheers, Kev

flackpoint, 2019-01-12:
Since there is no answer yet and i also struggled to get this running, here is my solution:

1. Create a gulp task for processing your source files (Template: [Optimizing files for SPIFFS with Gulp](http://tinkerman.cat/optimizing-files-for-spiffs-with-gulp/#lightbox-gallery-gNMBzR41/1/) )
2. Create a build task in vscode which contains these to parameters:
> "command": "${command:esp8266fs.packSpiffs}",
> "dependsOn": ["yourgulptask"]

This will first execute your gulp task and after that execute the packSpiffs command.

This is my tasks.json file:
```json
{
    "version": "2.0.0",
    "tasks": [
        {
	    "label": "Build SPIFFS Image",
            "command": "${command:esp8266fs.packSpiffs}",
            "group":{
                "kind": "build",
                "isDefault": true
            },
	    "presentation": {
                "panel": "dedicated",
                "reveal": "never"
	    },
	    "dependsOn": [
                "buildfs2"
	    ]
	},
        {
            "label": "buildfs2",
            "task": "buildfs2",
            "type": "gulp",
            "problemMatcher": [
                "$gulp-tsc"
            ],
            "group": "build"
        }
    ]
}
```

References:
[Run commands from tasks in vscode](https://code.visualstudio.com/docs/editor/variables-reference#_settings-command-variables-and-input-variables)