kash4kev/vscode-esp8266fs #10: ESP8266 SPIFFS interface started: "upload". Can't determine FlashSize.

thearduinoguy, 2019-04-12:
Attempting to upload to SPIFFS on a Wemos D1 Mini and I keep getting this message:

ESP8266 SPIFFS interface started: "upload".
Can't determine FlashSize.

I cannot for the lfie of me figure out what I need to do to fix this. Please help. Thanks!

maciekn, 2019-04-13:
@thearduinoguy did you try by any chance to check out fixed version from my PR? https://github.com/kash4kev/vscode-esp8266fs/pull/9
I was able to make it working on my machine

chartinger, 2019-05-28:
Had similar problems. The changes from PR #9 and an additional fix to remove the double quotes from the command parameter of spawnSync made this extension work for me. Also: don't forget to set the FlashSize in the board config to a value with SPIFFS memory.