kash4kev/vscode-esp8266fs #12: Plugin stops after "Uploading SPIFFS file with esptool.py"

FordPrfkt, 2019-08-18:
The plugin stops after:
`Output Port: COM4
mkspiffs: C:\Users\danie\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9\mkspiffs.exe
Found "mkspiffs" via VSCode Configuration
esptool: C:\Users\danie\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\tools\esptool\esptool.py
Found "esptool" via VSCode Configuration
--- Uploading SPIFFS file with esptool.py ---
Python Executable: "python" 
`
Nothing happens and vscode tells me the plugin is still running if i try to restart it

maciekn, 2020-01-06:
@FordPrfkt I think I have encountered similar issue recently and fixed in on my branch @ https://github.com/maciekn/vscode-esp8266fs/tree/issue/missing-esptool

As far as I have found out, the overarching reason is breaking change in ESP version 2.5 with the way how esptool is being managed. More details at https://github.com/esp8266/Arduino/pull/5635/files

As I am also attaching compiled extension from that branch: 
[vscode-esp8266fs-1.1.1-maciekn-snapshot.vsix.zip](https://github.com/kash4kev/vscode-esp8266fs/files/4024285/vscode-esp8266fs-1.1.1-maciekn-snapshot.vsix.zip)

InnuendoPi, 2020-01-07:
Thx! Works. 
quick n dirty I added + "m" to flashFreq and replaced const flashSize with "detect"
Innu

maciekn, 2020-01-07:
FWIW, after I spent few hours trying to fix SPIFFS upload from VS Studio, I found PlatformIO which solved this challenge* in a painless way. I highly recommend trying this one if you haven't done it yet!

*and few other challenges I had with Arduino addon for ESP8266 like incremental compilation