kash4kev/vscode-esp8266fs #13: Can't determine FlashSize.

ysg1998, 2019-12-07:


maciekn, 2019-12-08:
Looks like duplicate of #10

InnuendoPi, 2020-01-02:
Not working for LOLIN(WEMOS) D1 R2 & mini with ESP8266 2.6.3, ESP8266fs 0.5.0 and fix #9
Arduino IDE 1.8.10, Visual Code 1.41.1

`ESP8266 SPIFFS interface started: "upload".
Found ESP8266 packages: C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3
Found "esptool" via VSCode Configuration
--- Uploading SPIFFS file with esptool.py ---
SPIFFS Uploading Image... (C:\Arduino\xxx\build\xxx.spiffs.bin)
  [SPIFFS] Python   : python
  [SPIFFS] EspTool  : C:/Users/xxx/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py
  [SPIFFS] address  : 0x100000
  [SPIFFS] port     : COM3
  [SPIFFS] speed    : 460800
  [SPIFFS] before   : default_reset
  [SPIFFS] after    : hard_reset
  [SPIFFS] flashMode: qio
  [SPIFFS] flashFreq: 40m
  **[SPIFFS] flashSize: 4M3M**

usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument --flash_size/-fs: **4M3M is not a known flash size**. Known sizes: 512KB, 256KB, 1MB, 2MB, 4MB, 2MB-c1, 4MB-c1, 8MB, 16MB, detect, keep

python returned 2
ESP8266 SPIFFS interface finished.`

maciekn, 2020-01-06:
@InnuendoPi it looks like we can try to jump over the problem instead of solving it. In the meantime distribution of esptool has changed and it looks like it is easier to just not pass it. At least "official" Arduino SPIFFS addon is not doing it (https://github.com/esp8266/arduino-esp8266fs-plugin/blob/56a6c515545ff5301be5e679d014ab7062600d81/src/ESP8266FS.java#L334)

You may want to try my fixed version of extension as it is working for my case: https://github.com/kash4kev/vscode-esp8266fs/issues/12#issuecomment-570963519 (YMMV, but I am more than happy to make sure it will work for both if us)